usdot-fhwa-stol/carma-platform #432: Develop world model for external objects (perception)

msmcconnell, 2019-11-26:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
To safely plan, guidance will need access to a common interface which describes static and moving objects in the world around it. This interface needs to be defined and implemented. 
### Expected behavior
Guidance has mechanism to detect and avoid collisions
### Actual behavior
Logic existed in CARMA2 but needs to be ported/re-implemented in c++ for carma3.

maefromm, 2019-12-17:
@msmcconnell can this be closed?

msmcconnell, 2020-02-24:
@maefromm Not yet the external objects interface has only recently been started and has a fair amount left to do.

closed 2020-05-05 by TonysCousin