usdot-fhwa-stol/carma-platform #448: Autoware plugin's plan_trajectory response with empty plan

qswawrq, 2019-11-29:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Autoware plugin does not populate trajectory plan in plan_trajectory service response.
### Expected behavior
Autoware plugin response proper trajectory plan in plan_trajectory service response.
### Software version this applies to

3.1.0
### Actual behavior
Autoware plugin does not populate trajectory plan in plan_trajectory service response.
### Steps to reproduce the behavior
Make service call to Autoware plan_trajectory service. It will return an empty plan.
### Related work
#410

qswawrq, 2019-11-29:
After investigation, it is not an anomaly since it responds with empty trajectory when waypoints are not loaded. Closed.

closed 2019-11-29 by qswawrq

maefromm, 2019-12-17:
As discussed there was no changes and non-issue.