usdot-fhwa-stol/carma-platform #459: Vehicle does not drive the route despite trajectories being generated

Supraluminal, 2019-12-12:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
After resolving issue #451 the Arbitrator now properly emits maneuver plans at the expected 1hz rate. However, despite this, the vehicle does not begin motion along the route. Maneuver plans are being generated and converted into trajectories in a continuous fashion as evidenced by the Trajectory Executor not commanding a shutdown when it runs out of trajectory data. The issue likely lies somewhere in the interaction between the Plan Delegator -> Trajectory Executor -> Pure Pursuit Wrapper, but it is not immediately clear where the issue lies.

### Expected behavior

After pressing engage the vehicle should begin motion and drive the configured route. Generated maneuver plans on `/guidance/final_maneuver_plan` should be converted into trajectories by `plan_delegator` which are then executed by `pure_pursuit_wrapper` by sending commands into the SSC controller interface.

### Software version this applies to
Most recent tag is CARMA 3.1.0 but the latest fixes from #451 are soon to be on develop as a result of #458 .

### Actual behavior

Vehicle remains stationary and does not begin driving. Maneuver plans are generated and trajectories are generated but the vehicle does not drive.

### Steps to reproduce the behavior

1. Run `carma start all`
2. Localize the vehicle in RVIZ
3. Select TFHRC Primary as route
4. Enable Autoware Plugin and Autoware Plugin View
5. Engage Guidance

### Related work
#451 
#458

qswawrq, 2019-12-12:
It seems to be a false positive. It is resolved in the latest develop already. I suspect the docker image you built on the car does not contain those fixes.

qswawrq, 2019-12-12:
verified on Fusion. Closed.

closed 2019-12-12 by qswawrq