usdot-fhwa-stol/carma-platform #469: Issue with Ford Fusion: Guidance re-engage capability so that multiple runs can be made without restarting the software

aswath1, 2019-12-17:
### Types of Issue

- [x ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Issue with Ford Fusion: Guidance re-engage capability so that multiple runs can be made without restarting the software
### Expected behavior
Multiple runs can be made without restarting the software.
### Software version this applies to

### Actual behavior
Needs  restarting the software.
### Steps to reproduce the behavior

1. Do this
1. Then do this...

### Related work

Link to related tickets or prior related work here.

aswath1, 2019-12-22:
The issue is fixed through PR : release/testarossa - component fix for trajectory error and ensuring docker id is 999 #480

closed 2019-12-22 by aswath1

maefromm, 2019-12-23:
Ash removed from this issue from the testarossa test plan since duplicate issue with https://github.com/usdot-fhwa-stol/CARMAPlatform/issues/476. This is fixed in the 476 issue.