usdot-fhwa-stol/carma-platform #473: CARMAWebUI for testarossa release doesn't show version testarossa on the Web UI

aswath1, 2019-12-18:
### Types of Issue

- [x ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
CARMAWebUI for testarossa release doesn't show version testarossa on the Web UI
### Expected behavior
The web ui should show the version as testarossa.
### Software version this applies to

### Actual behavior
The web ui doesn't show the version as testarossa.
### Steps to reproduce the behavior

1. Do this
1. Then do this...

### Related work

Link to related tickets or prior related work here.

maefromm, 2019-12-19:
This issue only occured in the Blue Lexus and Pacifica. Ford Fusion worked. 

The issue was caused due to the docker group id in the vehicle host PC and web-ui image is using different IDs, which didn't allow it to execute the docker inspect command. Eventhough we already have the docker:www-data in the etc/groups in both the image and the host PC, it was not enough.  The group # is 999  in the image and 998 on the PC. Switching the group id is a more work and can be complicated. 

At the meantime, the work around is to issue this command "sudo chmod 777 /var/run/docker.sock" on the vehicle PC, which allows everyone to access the docker.sock file. 

More information explained in http://blog.gaiterjones.com/docker-in-docker-permissions/

Screenshots below: 
[image]
[image]

maefromm, 2019-12-19:
Created a startup script that will need to be manually installed if the group id does not match with the container's docker group id=999. Will check in under engineering tools.

aswath1, 2019-12-22:
The issue is fixed through PR : release/testarossa - component fix for trajectory error and ensuring docker id is 999 #480

closed 2019-12-22 by aswath1