usdot-fhwa-stol/carma-platform #475: Vector Map and Point Cloud Map transformation does not match

itabrl, 2019-12-18:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
vector map and point cloud map visualizations in rviz does not over lap, vector map is not aligned  by some degree. some of the lanes are going through walls.
### Expected behavior
vector map to be aligned with point cloud map
### Software version this applies to

develop
### Actual behavior

### Steps to reproduce the behavior

1. load vector map and point cloud map into Rviz

### Related work

Link to related tickets or prior related work here.

msmcconnell, 2020-01-24:
This was found to be caused by a incorrect conversion of the mapping companies internal LAS format to PCD. We are in the process of verifying corrected PCD files in our other maps

itabr, 2020-02-05:
the issue was verified to be an incorrect conversion of the mapping companies' internal LAS format to PCD. the updated maps are now verified to have a correct transformation.

maefromm, 2020-02-05:
Closing this issue as we received the updated files from ContinentalMapping. At this time only validated TFHRC, Perryman, ATEF and SummitPoint. The McLean, I-95 and 495 has not been revalidated, and will be done later when we get updated semantic map updates for McLean, I-95 and 495. If there are any issues on those, we can open a new ticket for each map. Thanks!

closed 2020-02-05 by maefromm