usdot-fhwa-stol/carma-platform #482: CARMAAvtVimbaDriver Build Image Name is Incorrect

maefromm, 2019-12-21:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Found this issue during the build-image.sh of this repo. 

### Expected behavior
should have dashes in the image name. 

### Software version this applies to

testarossa

### Actual behavior

### Steps to reproduce the behavior

1. Do this
1. Then do this...

### Related work

Link to related tickets or prior related work here.

aswath1, 2019-12-22:
Fix PR: fix build image name #25

closed 2019-12-22 by aswath1

maefromm, 2019-12-23:
@jtbaird has confirmed the build I pushed is out on the right repo.