usdot-fhwa-stol/carma-platform #483: Ford Fusion: Intermittent and potential HW connection issues

maefromm, 2019-12-23:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During the testarossa release testing, experienced the following INTERMITTENT issues:
1. The PC is not able to bootup. The system is getting powered but we can't see any ubuntu bootup neither we can ssh into the system. 
2. One of the sensors failed during a drive and Engineer thinks it was a connection issue that's loose. 

Note that issue is intermittent and the system worked after a while. The Testarossa release was successful and tested on Fusion. 

### Expected behavior
PC should always boot up. 

### Software version this applies to

N/A

### Actual behavior
Intermittently not booting up the PC and some connection issues with a driver. 

### Steps to reproduce the behavior
Intermittent after booting up the system. 

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-01-14:
@aswath1 reported discovering and fixing some loose cable connectors on the spectra computer several days ago.  Let's keep that in mind if we discover more of this behavior, in case they come loose again.

TonysCousin, 2020-04-17:
No reports lately of this recurring, but it could come back intermittently.  Therefore, assigning sev 3.

aswath1, 2020-04-17:
Yes