usdot-fhwa-stol/carma-platform #491: CARMAGarminLidarLiteV3DriverWrapper should publish a message type containing header in trailer_angle topic

msmcconnell, 2020-01-02:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The trailer angle reported by the CARMAGarminLidarLiteV3DriverWrapper is published as an std_msgs/Float64 message type. It is difficult to use this message type reliably as it does not contain a timestamp which can be used for transform lookups. It should be updated to use a message type containing a header. 
### Expected behavior
trailer_angle reported with timestamp
### Software version this applies to

CARMASystem_3.2.0
### Actual behavior
No timestamp reported
### Steps to reproduce the behavior
Look at source code.

aswath1, 2020-02-09:
fixed under https://github.com/usdot-fhwa-stol/CARMAGarminLidarLiteV3DriverWrapper/pull/12

closed 2020-02-09 by aswath1

msmcconnell, 2020-02-10:
@aswath1 don't close the issue until the PRs have been merged