usdot-fhwa-stol/carma-platform #497: PACMOD and/or SSC module status messages appear to have different semantics on truck

Supraluminal, 2020-01-14:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The Red Freightliner Cascadia appears to have slightly different semantics for the states of the PACMOD/SSC interface modules. This results in the CARMAPlatform being unable to properly startup as our drivers never report a ready state.
### Expected behavior

1. Start CARMA via `carma start all`
1. SSCInterfaceWrapper receives ready module states from PACMOD/SSC
1. SSCInterfaceWrapper reports `DRIVER_READY` to `health_checker` node.
1. CARMA Platform successfully starts up.

### Software version this applies to

CARMASystem_3.2.0, modified to run on the truck as shown on the develop branch of the CARMAConfig repo.

### Actual behavior

1. Start CARMA via `carma start all`
1. SSCInterfaceWrapper receives driver not ready status from SSC modules
1. SSCInterfaceWrapper reports `OFF` to `health_checker` node.
1. CARMA Platform shuts down on the command of the `health_checker` node

### Steps to reproduce the behavior

1. Deploy the truck config
1. Run `carma start all`

maefromm, 2020-01-17:
AStuff support ticket 2688 has been forwarded to Saina and team.

maefromm, 2020-01-24:
@msmcconnell please kindly close the AStuff ticket with your final findings and resolution. And close this Github issue when merged your PR changes. Thanks!

closed 2020-01-24 by qswawrq