usdot-fhwa-stol/carma-platform #499: Trailer angle sensor devices are not mapped on the host or in docker

Supraluminal, 2020-01-14:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The trailer angle sensors deployed on the freightliner cascadia attempt to read from devices `/dev/ttyACM0` and `/dev/ttyACM1`. These devices do not appeared to be configured on the host, or mapped in through docker, resulting in read errors when running the trailer angle sensor driver.
### Expected behavior

1. Start CARMA via `carma start all`
1. The Trailer Angle Sensor driver is able to read the devices and process the data
1. The Trailer Angle Sensor outputs data via it's ROS topics.

### Software version this applies to

CARMASystem_3.2.0, modified to run on the truck as shown on the develop branch of the CARMAConfig repo.

### Actual behavior

1. Start CARMA via `carma start all`
1. The Trailer Angle Sensor driver is unable to read the devices and process the data
1. The Trailer Angle Sensor does not output data via it's ROS topics.

### Steps to reproduce the behavior

1. Deploy the truck config
1. Run `carma start all`

maefromm, 2020-01-15:
As discussed during the sprint 23 planning this could be considered both anomaly or enhancement depending on the reason behind the issue. For now updating to enhancement.

aswath1, 2020-01-30:
Solved by https://github.com/usdot-fhwa-stol/CARMAConfig/pull/29

closed 2020-01-30 by aswath1