usdot-fhwa-stol/carma-platform #501: Adding Routing Mechanism

qswawrq, 2020-01-14:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
We want route generator to actually generate a route based on user's input.
### Expected behavior
Route generator generate a route based on user selected destination point.
### Software version this applies to

N/A
### Actual behavior
N/A
### Steps to reproduce the behavior
N/A

### Related work
N/A
Link to related tickets or prior related work here.

maefromm, 2020-01-15:
@qswawrq please make sure you follow the new design we discussed, I thought we plan to place everything into one package/node?

qswawrq, 2020-01-30:
Closed by #502

closed 2020-01-30 by qswawrq