usdot-fhwa-stol/carma-platform #504: TFHRC LaneLet Vector Map Does Not Contain Correct Lanelet Connections

qswawrq, 2020-01-17:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Using lanelet2 converter script, we can convert OpenDrive vector map to OSM format. However, converted map cannot be used to generate correct route graph. For example, in TFHRC vector map, we have graph: 
[routing.txt](https://github.com/usdot-fhwa-stol/CARMAPlatform/files/4074960/routing.txt)
, which can not be used to routing due to missing relations between lanelets.
### Expected behavior
Vector map can be used for routing.
### Software version this applies to

3.2.0
### Actual behavior
Converted vector map can be used for routing at current version.
### Steps to reproduce the behavior
Use lanelet2 lib to route on TFHRC vector map.
### Related work
N/A
Link to related tickets or prior related work here.

msmcconnell, 2020-01-17:
At the moment I think this is expected as Jira task CAR-1533 has not bee completed yet.

qswawrq, 2020-01-17:
> At the moment I think this is expected as Jira task CAR-1533 has not bee completed yet.

Then this ticket should be marked as enhancement.

qswawrq, 2020-01-17:
Confirmed in ATEF vector map that this issue still exist.

qswawrq, 2020-02-05:
@msmcconnell Should this be closed since CAR-1533 is merge?

msmcconnell, 2020-04-17:
@qswawrq I believe this could be closed given that we have been able to properly route on the map now.

qswawrq, 2020-05-05:
Implemented by #508. Closed.

closed 2020-05-05 by qswawrq