usdot-fhwa-stol/carma-platform #517: vector map to be loaded when carma is launched

itabrl, 2020-01-24:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [X] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
After running carma start all, vector map needs to get loaded. 

### Expected behavior

### Software version this applies to

3.2.0
### Actual behavior
vector map does not get loaded
### Steps to reproduce the behavior

1. Do this
1. Then do this...

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-01-29:
This was fixed under https://github.com/usdot-fhwa-stol/autoware.ai/pull/45

maefromm, 2020-01-29:
Also https://github.com/usdot-fhwa-stol/autoware.ai/pull/55

maefromm, 2020-01-29:
@itabrl please include fix for rviz profile to include "LaneLet2" display.

closed 2020-02-05 by maefromm