usdot-fhwa-stol/carma-platform #531: Add automated package.xml version updates to release process

jtbaird, 2020-01-29:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Currently process to update package.xml versions for a release is entirely manual. An automated process could update these files faster, more accurately, and ensure that anything dependent on that package version is likewise updated.

### Expected behavior
**For a major component release:**
- update all package.xml versions to match new component version
- if an updated package is referenced in other repo, update version there as well
- output package.xml name, old version, new version to .csv
- run `git add .`, then make a commit, and add the commit has to the .csv to track the version change

**For non-major component release:**
- check parent dir of each package.xml for changes in git since last release
- if there have been changes, update version as minor or bugfix depending on git info
- if an updated package is referenced in other repo, update version there as well
- output package.xml name, old version, new version to .csv
- run `git add .`, then make a commit, and add the commit has to the .csv to track the version

**For system releases:**
- same as non-major component release, but across all repos involved in release

### Software version this applies to
All.

### Actual behavior
Process is entirely manual.

### Steps to reproduce the behavior
Many.

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-01-31:
@jtbaird a couple clarifications are in order:
1) for a non-major component release (i.e. defect fixes only, that would affect the 3rd value in the version ID?) is the intent only to update on the package.xml of the package that was changed, and not all the package.xmls in the repo?
2) We had previously discussed not touching any package.xml in the forked repos. Does this proposal now apply to those repos as well?  How would that impact future syncs with their origin or pushes back to their origin?
3) Presumably this automation script would live in engineering_tools, is that right?

jtbaird, 2020-01-31:
1. Correct, this way the package.xml version would accurately reflect the state of the package.
2. I was planning to leave the forked versions alone for now.
3. engineering_tools seems like a good location for the time being. Ideally though, these scripts would exist in a local repo on a dedicated build machine (cloud or on-prem), this would also allow for further release orchestration since it would provide us a place to store information about dependencies between repos. Without that type of structure, it's not totally clear how we should go about coordinating this across multiple repositories.

TonysCousin, 2020-01-31:
Sounds good.  Go for it.

closed 2020-04-17 by jtbaird