usdot-fhwa-stol/carma-platform #536: Red Truck - Unable to re-engage after override?

maefromm, 2020-02-04:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Using Autoware on the red truck, we found that we are unable to re-engage after the driver manually overrides using brake/steering/throttle/e-stop button. 

### Expected behavior
Similar passenger vehicles, we should be able to re-engage Autoware without having to restart the entire system and having to press the e-stop button. 

### Software version this applies to

Autoware 1.12/1.13

### Actual behavior
The re-engage the system, we are required to fully stop Autoware and press the e-stop button. 

### Steps to reproduce the behavior
1.      Start Autoware  and send the vehicle/engage true command. 
2.      After driver did a brake override, sent the vehicle/engage false command.
3.      Shift back from Drive to Neutral
4.      Send the vehicle/engage false command.
5.      Send the vehicle/engage true command.

### Related work

Link to related tickets or prior related work here.
Opened AutonomouStuff Ticket# 2667

TonysCousin, 2020-02-27:
Blue, silver & white do not exhibit this problem.

TonysCousin, 2020-04-17:
Removing anomaly label, as it appears to be a problem with Autoware, which is no longer a concern since we are now installing Carma on the trucks instead.  @maefromm need an update to confirm whether this is just an Autoware problem, or something bigger.

maefromm, 2020-04-17:
@msmcconnell have we retested the re-engage on red truck since the last time we were out on ATEF? I believe this was related to an SSC software update.