usdot-fhwa-stol/carma-platform #540: Red Truck - PACMOD lighting strips in trucks

maefromm, 2020-02-04:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The PACMOD lighting strips in the trucks can be blinding at night. Currently there is a switch that controls the dim levels for both the LED lights and the dashboard lights. 

Sent the picture to Ben today that the dimmer switch (left of the hazards button) is no good, since it affects the dashboards lights too. So if you dim the LED lights to low setting, it also dims the dashboard lights and you can’t drive without that at night.

### Expected behavior

### Software version this applies to

### Actual behavior

### Steps to reproduce the behavior

1. In a dark room, test the dim levels for the LED lights and ensure it doesn't effect the dashboard lights for safe driving. 

### Related work

Link to related tickets or prior related work here.
ASTuff ticket 2748

maefromm, 2020-02-04:
[RE AutonomouStuff Re RE EXTERNAL AutonomouStuff Re Re Pacmod lighting strips in trucks.txt](https://github.com/usdot-fhwa-stol/CARMAPlatform/files/4154758/RE.AutonomouStuff.Re.RE.EXTERNAL.AutonomouStuff.Re.Re.Pacmod.lighting.strips.in.trucks.txt)

TonysCousin, 2020-11-02:
This issue was first noticed on the red truck, but it is going to be the same on all of the trucks, so removing truck-specific wording from the title.

msmcconnell, 2021-03-01:
As far as I know this was resolved by AStuff. Closing

closed 2021-03-01 by msmcconnell