usdot-fhwa-stol/carma-platform #543: Blue Lexus leaves road during waypoint following on Urus release candidate

Supraluminal, 2020-02-05:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The Blue Lexus begins to drive off of the road during CARMA waypoint following demo using Urus Release Candidate build. The vehicle travels the TFHRCPrimaryRoute from the C-Hub to shortly before the first intersection when it appears to understeer out of the second bend of the s-curve and heads off the paved surface, requiring safety driver override. This behavior is observed with the old map and route which should be aligned in the same coordinate system. The Pacifica using the same map and it's equivalent route for that map does not display this same behavior.
### Expected behavior
Lexus stays within legal lane markings throughout entire route.
### Software version this applies to

### Actual behavior
Lexus leaves lane on the right side of the road shortly before TFHRC first intersection.

### Steps to reproduce the behavior

1. Deploy `CARMASystem_Urus...lexus` CARMAConfig
2. Load old TFHRC PCD map and ensure that TFHRCPrimaryRoute is available
3. Start carma and engage the TFHRCPrimaryRoute

maefromm, 2020-02-07:
Saina has completed her testing with the Ford and Silver Lexus using the Urus release, and it worked. So Red Truck, Pacifica, Ford, and Silver Truck works with Urus release. 

The issue with the Blue Lexus losing localization and going off road seems to be particular to this vehicle only, and not related to the Urus release changes. Therefore we moved down the severity of Issue Github Issue #543 from 1 to 2. We will have to further investigate the issue.

saina-ramyar, 2020-02-07:
It seems like the localization on the blue Lexus is unstable, and that is causing the vehicle to go off the road. Localization is not completely lost but it does not lock down either and drift a little. As mentioned in #544 the Blue Lexus worked well with 3.2.0 and does not lose localization.

msmcconnell, 2021-03-01:
Urus release was truck specific and we will not be going back to investigate this further

closed 2021-03-01 by msmcconnell