usdot-fhwa-stol/carma-platform #549: Route file naming supports all valid linux file names and route ids are unique

msmcconnell, 2020-02-06:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The waypoint_loader from autoware currently does not support file paths with spaces in them. Once the waypoint_loader is removed as part of routing improvements the route file loading mechanism should be confirmed to support all valid routes in the linux file system. In addition, the routeID used to communicate the current route within the system should be unique (alpha/numeric) and independent of the route file name. 
### Expected behavior
All valid file names in linux are valid route file names. 
### Software version this applies to

CARMA Urus release
### Actual behavior
Routes with spaces in file name are not supported
### Steps to reproduce the behavior

Try loading route with space in file name. 

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-02-06:
Filenaming convention for our route files for the new route node I think should be:

1. Case insensitive like the filename. Mix of upper case and lower case allowed. Which means filenames "MyFile.csv" and "myfile.csv" would be considered different files and be allowed to load. 
2. Filenames allows special characters but I think we should limit it to such as blank space, hyphens and underscores. Not sure about the dots, %, $ and others. If the RouteID is updated to be non-space and just alphanumeric, UI display of special charters on the value of the label/button shouldn't matter but will need to double check on this.
3. Keep filenames short and descriptive. Not sure what the maximum character linux allows. If none then still need to limit it since UI will not display it all properly.

TonysCousin, 2020-04-17:
Closed as resolved.

msmcconnell, 2020-09-22:
Re-opening as this was never fixed only a log statement was added.