usdot-fhwa-stol/carma-platform #555: Register all lidars with health monitor

TonysCousin, 2020-02-10:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
On multi-lidar vehicles (the trucks currently have 2 lidars) only one of them is registered with the health monitor. If it dies health monitor will raise a system alert, notifying that a critical sensor is gone, which will result in system shutdown.  However, having two functionally identical lidars on the truck, we may not want to consider loss of one as a system critical failure.  The truck could still operate with the other lidar working.  Secondly, as currently configured, if the non-monitored lidar fails, health monitor will not know anything about it, and thus will not report the condition at all.  It needs to be monitoring both lidars, and issue an appropriate level of alert based on detection of either one or both lidars failing.

### Expected behavior
If either lidar on the truck fails, then health monitor issues a warning. If both have failed then it issues a critical failure message.

### Software version this applies to

3.3.0

### Actual behavior
If lidar #1 fails, then critical failure message is issued (system shutdown). If lidar #2 fails, then it is not detected and there is no reaction.

### Steps to reproduce the behavior

1. Operate a Carma truck.
1. Power down one of the lidars.

### Related work
None.

msmcconnell, 2020-03-11:
This was implemented prior to Urus release in 
usdot-fhwa-stol/CARMAConfig#31

TonysCousin, 2020-04-17:
Health monitor now issues correct level of alert based on either or both of the lidars showing a fault.

closed 2020-04-17 by TonysCousin