usdot-fhwa-stol/carma-platform #557: Load map transforms from OSM file.

msmcconnell, 2020-02-10:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
To simplify usage of HD maps in carma we should load the open drive proj georeference directly from the loaded osm file and use this as the ecef to map transform
### Expected behavior
Singular source of map to ecef transforms and projections. 
### Software version this applies to

Urus release
### Actual behavior
Both the map projection and map to ECEF transform are loaded separately.

msmcconnell, 2020-03-11:
This is effectively a duplicate issue of #360

TonysCousin, 2020-04-17:
Closing per Mike McConnell.

closed 2020-04-17 by TonysCousin