usdot-fhwa-stol/carma-platform #568: Wheel base value inconsistent on red truck config

msmcconnell, 2020-02-18:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The vehicle_wheel_base parameter found in CARMAConfig here
https://github.com/usdot-fhwa-stol/CARMAConfig/blob/develop/freightliner_cascadia_2012/VehicleConfigParams.yaml
(4.525m)
Does not match the value used in the SSC found here
https://github.com/usdot-fhwa-stol/CARMAVehicleCalibration/blob/develop/2012-freightliner_cascadia-DOT_10003/vehicle/calibration/ssc_pm_cascadia/json/steering_model.json
(5.334m)
Having inconsistent values could lead to poor steering control behavior. 
### Expected behavior
### Software version this applies to

CARMASystem_3.3.0

maefromm, 2020-02-18:
@msmcconnell as discussed the "the wheel base  in this case is an approximation due to multiple rear axels." The Json does not actually matches the actual wheel base that Ed provided for each truck, correct? If so, I have already approved PR,, feel free to merge.

msmcconnell, 2020-02-19:
@maefromm Correct. I will merge the fix

closed 2020-02-19 by msmcconnell