usdot-fhwa-stol/carma-platform #588: CARMA 3.3.0 - Initialization/Localization Issue

maefromm, 2020-03-04:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

There are 2 issues involved with Initialization/Localization on CARMA 3.3.0. 

1) Manual initialization/localization has become more tedious and very rigid, and sometime takes a while to snap to the map. 

Compared with 3.2.0, it seems that the initialization/localization shows that after manual localization, it actually continuous to try to snap onto the map. 

Per discussion with @msmcconnell, this seems to be related to the change of removing of voxel grid filter in CARMA 3.3.0 and only using ray ground filter. Autoware seems to be only using voxel grid filter and not ray ground filter. Mike thinks switching to voxel will improve initialization/localization. 

2) Automatic Initialization/Localization by CARMA 3.3.0 is about 2 meters away and doesn't snap to the map automatically which may be related to #1 or may be fixed by it. @msmcconnell regardless of Micheel's fix to the map origin, the map's GPS coordinate being different with the vehicles GPS coordinates still may be the issue. 

### Expected behavior

### Software version this applies to

CARMA 3.3.0 

### Actual behavior
Described above. 

### Steps to reproduce the behavior
1. Run CARMA 3.3.0 using the CARMA3 desktop icon on the vehicle 
2. Notice that RVIZ doesn't snap to the map automatically, it may be close but never localizes. 
3. Notice that when you start moving to try to localize, it gets even worse. 
4. When trying to manually localize using command line and RVIZ, it is very difficult and very rigid. 

### Related work

Link to related tickets or prior related work here.
Other issues related to this  in the past that has been close are : #273 #384

TonysCousin, 2020-04-16:
Assigning this as sev 2 because it is a mission critical function, but there is a workaround in doing a manual alignment, however tedious it may be.

msmcconnell, 2020-07-24:
This has been resolved by #767

closed 2020-07-24 by msmcconnell