usdot-fhwa-stol/carma-platform #592: CARMA UI - Update to handle new Route Manager

maefromm, 2020-03-06:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
With the new Route Manager, need to update that CARMA UI to use the latest topics, services, route events, and states. 
Related JIRA Task 1558

### Expected behavior
At minimum work with the new Route node. 

NOTE: Additional changes need to be made to UI to address the obsolete topics from CARMA2 or new topics to accomodate the new design for CARMA3. This will be separate tickets. 

### Software version this applies to

3.3.0

### Actual behavior
See above. 

### Steps to reproduce the behavior
Run CARMAPlatform. 
Open UI and should be able to work with the new implementation of Route. 

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-05-18:
The code has been completed under the PRs above, however additional changes may occur during integration testing. Will leave this open for now until then.

msmcconnell, 2021-03-01:
Resolved by above PRs and being handled under separate issues for integration testing.

closed 2021-03-01 by msmcconnell