usdot-fhwa-stol/carma-platform #598: Truck OBUs need to be reconfigured

TonysCousin, 2020-03-09:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During acceptance testing of the blue, white & silver trucks at AStuff, it was observed that one of the OBUs in blue is configured as an RSU (which provides message forwarding capability), but its other OBU, and both OBUs in the white & silver trucks, are configured as an OBU.  More details of the individual unit configurations are not clear, but they should all be set up with the RSU firmware and then configured with the message forwarding and disabling of WSA generation.

### Expected behavior
All OBUs are configured as RSUs with message forwarding capability.

### Software version this applies to

N/A

### Actual behavior
Most truck OBUs are configured as OBUs.  This makes message forwarding much more difficult (requires new software to be written).

### Steps to reproduce the behavior

1. ssh into the OBU
2. Credentials in OBU configuration will be "user"; credentials in the RSU config will be "rsu".

abey-yoseph, 2020-09-02:
This issue should be resolved. The OBUs have been upgraded to run the “mk5-17.Release.103771-RSU1609-typical.img” firmware that is available on the Cohda box website. This firmware has the OBU run as an RSU, which automatically enables the “immediate-forward” capability. This will automatically broadcast messages sent over port 1516 out. To enable BSM forwarding (this is the opposite functionality…the OBU receives a BSM over the air and forwards it to the spectra), the attached “rsu_setup..” script is placed in the /mnt/src/ directory of the OBU. The attached “rc.local” file is placed in the /mnt/rw/ directory of the OBU. The rc local will call the rsu_setup script upon startup. This script will enable BSMs to be forwarded to the spectra (192.168.88.10). The rsu_setup script is configurable, so if we want to add other J2735 messages to be forwarded to the spectra, we can.

abey-yoseph, 2020-09-02:
[rc.local.txt](https://github.com/usdot-fhwa-stol/carma-platform/files/5164151/rc.local.txt)
[rsu1609_setup.20200808.sh.txt](https://github.com/usdot-fhwa-stol/carma-platform/files/5164153/rsu1609_setup.20200808.sh.txt)

abey-yoseph, 2020-09-03:
[rsu1609_setup.20200808.sh.txt](https://github.com/usdot-fhwa-stol/carma-platform/files/5169287/rsu1609_setup.20200808.sh.txt)

msmcconnell, 2021-03-01:
Resolved per above notes

closed 2021-03-01 by msmcconnell