usdot-fhwa-stol/carma-platform #608: vector_map not loading correctly

MishkaMN, 2020-03-18:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
vector_map is not loading correctly when carma starts even though geoReference is inside vector_map.osm. 
### Expected behavior
- run `carma start all`
- in `rviz`, lanelet should be there
- map_param_loader node should be up and running
- `rostopic echo tf_static` should have ECEF points (large numbers) in its transform with `earth` frame id.
### Software version this applies to

### Actual behavior
- lanelet is not there in `rviz`
- `rostopic echo tf_static` has [0 0 0 1] component in `earth` frame, which should not be happening.

### Steps to reproduce the behavior

Outlined above. 

### Related work

Link to related tickets or prior related work here.

MishkaMN, 2020-03-18:
One possible reason might be that the `geoReference` is not correctly added to the vector_map.osm
It should be in the following format:
`<geoReference>georeference_string</geoReference>`
If the map_param_loader cannot find this `geoReference` tag, it should throw an error. However if the tag is there, but incorrectly encoded such as this:
`<geoReference v="georeference_string"/>`
it will still find the geoReference, so will not throw error while vector_map is not loading correctly.
When parsing, this scenario should also be handled.

msmcconnell, 2020-04-24:
Reopening as merged PR only logged an error rather than fixing the underlying issue.

TonysCousin, 2020-05-07:
Marking sev 2 as it affects mission-critical function (can't load a map that is the basis of all navigation). Discussion w/ @itabrl and @msmcconnell confirm that a workaround is available, by editing the individual map files to adjust their georeference info.

closed 2020-05-08 by MishkaMN