usdot-fhwa-stol/carma-platform #609: Silver and Pacifica Windows PC auto shutdown due to power issue

MishkaMN, 2020-03-18:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
This problem applies to Lexus and Pacifica.
When trying to login to the Windows PC (Crystal) using monitor, the computer automatically shuts down and reboots directly from the login screen. The issue is narrowed down to a power issue related to the PCD as bypassing PCD and directly supplying power from car to PC fixed the issue.

Currently, Silver uses external power generator supplied from the car to turn on Crystal.
Pacifica was rewired to get power from car to the Crystal.

As of now, it is not clear whether if the PC would have worked fine without trying to login using monitor through PCD. For example, mounting the PC when starts and successfully transferring data from Spectra Ubuntu.

### Expected behavior

Windows should not reboot automatically.

### Software version this applies to

N/A

### Actual behavior

Explained above.

### Steps to reproduce the behavior

On Silver and Pacifica,
1. Hook up monitor
2. Start the Crystal (powered from PCD)
3. Once the Windows Login screen starts, try clicking on `username` or `password` field, it will automatically reboot.

### Related work

MishkaMN, 2020-03-18:
From Astuff ticket:
From Crystal NAS documents, it looks like they require 18-36VDC.
12V from the PDS isn't going to supply enough.
Although this issue does not appear on the other 2 cars.

TonysCousin, 2020-04-22:
Comment from Ed Leslie on 4/21, "Reading back over Chris’s comments, our issue is not with voltage – we don’t have their Crystal units, ours is fine to run on 12V.  I’d still like to know from AStuff why it wouldn’t work through the PDS."

maefromm, 2020-10-05:
If no one else objects today, this issue can be closed. 
Per Ed: the issue occurred on the silver Lexus when it was wired to the PDS.  We started troubleshooting by connecting to a power supply and it worked.  Then we moved to the SMD block. I am ok keeping the NAS on the SMD block as long as that’s ok with the team.  The tradeoff is that the NAS can no longer be powered on or off from the PDS touchscreen.
Per John: I’m fine with that as long as it’s still on the master power switch.  It is rare that we will want to power on/off individual components.
Attached email chain
[Uploading RE NAS in Lexus .txt…]()

MishkaMN, 2020-10-05:
Closing this issue per comment left by Mae.
Thanks

closed 2020-10-05 by MishkaMN