usdot-fhwa-stol/carma-platform #610: RAID mount user mismatch on Pacifica and Fusion

MishkaMN, 2020-03-18:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Currently all carma log files are being written on RAID drives. As `docker` containers all have virtual user named `carma` with `uid=1000, gid=1000`, they all can only write to folders with those ids. 

Therefore, RAID's shared folder was mounted using that permission, specifically:
`sudo mount -t cifs -o username=<RAIDWindowsUser>,password=<RAIDWindowsPwd>,
uid=1000,gid=1000 //RAID_IP/WindowsSharedFolder /home/autoware/WindowsSharedFolder`

This works fine for `carma`'s functionality; however, when trying to debug with `rostopic` or `rviz` it gives `permission error` on all users except `uid=1000`. Therefore, since development primarily happens on `uid=1001 (autoware)` on Fusion and Pacifica, it creates an extra hassle.

**Current workaround** is to 
1. `unlink` the symlink on ubuntu:
(e.g. `/opt/carma/logs` -> `~/WindowsSharedFolder/logs` and for `.ros`)
2. or switch to user with uid=1000
3. or `docker exec` into the container to run `rostopic`
### Expected behavior

`rostopic` or `rviz` should work fine without any `permission error` while `carma start all` is running

### Software version this applies to
3.3.0 but not limited to it

### Actual behavior

Gives `permission error` on various logging commands

### Steps to reproduce the behavior

On Pacifica and Fusion,
1. Make sure the RAID is mounted on `~/WindowsSharedFolder`. 
2. Moreover, make sure `logs` and `.ros` folders are symlinked to it:
`/opt/carma/logs` -> `~/WindowsSharedFolder/logs` 
`/home/autoware/.ros` -> `~/WindowsSharedFolder/.ros`
3. run `carma start all`
4. `rostopic <any_topic>` it should give some permission error and will not print data in the terminal
 
### Related work

N/A

MishkaMN, 2020-03-18:
Perhaps for short term solution, if there is a way to `mount` the RAID with permission to multiple users, it might work. 
(without specifying `uid/gid` did not work as it defaults to `0`)

TonysCousin, 2020-04-16:
@MishkaMN it is worth testing whether this is really a user problem or just a group problem.  In our vehicle, the user "autoware" also has gid = 1001, right?  If we create a new user (say, with uid = 1078, but with gid=1000), then it might work.  We need to have a more robust OS environment that allows multiple users within the carma _group_ to use the full functionality of the system.

TonysCousin, 2020-04-16:
Also, I am changing this from an enhancement to an anomaly, because it describes a problem that needs to be worked around (Carma isn't performing as we would like it to).

MishkaMN, 2021-06-21:
This has been resolved by changing the main user name's id (either it is autoware or dev) into 1000 id to match that of docker's default id of 1000

closed 2021-06-21 by MishkaMN