usdot-fhwa-stol/carma-platform #624: Image can run but shut down immediately, on master branch.

Helomair, 2020-03-30:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
These errors occurs when running the docker image of master branch.
```
/home/carma/.base-image/entrypoint.sh: line 23: [: too many arguments
/bin/sh: 1: roslaunch carma carma_docker.launch: not found
```
And then the image shut down.

### Expected behavior
Docker image should be run locally.

### Software version this applies to

```
Ubuntu 19.10
Docker 19.03.06
```

### Steps to reproduce the behavior

1. Build image
```
docker build -t <name> . --no-cache
```
2. run & test it, but it shut down.
```
docker run -it <name>
```

msmcconnell, 2020-03-30:
@Helomair Thank you for submitting an issue. 
The docker image system used in carma platform is managed differently from just building the one image and running it. We use docker compose to control vehicle configurations comprised of multiple images. We are currently working on better documentation for this. I will try to give you some initial instructions below. 

1. Download and install the following two files manually
https://github.com/usdot-fhwa-stol/CARMAPlatform/blob/master/engineering_tools/carma
https://github.com/usdot-fhwa-stol/CARMAPlatform/blob/master/engineering_tools/__carma_autocomplete

The "carma" script will be used to manage your carma installation. Add "__carma_autocomplete" allows autocompletion to be used with the script. In Ubuntu 16.04 (currently our supported ubuntu version), these are placed in /usr/bin and /etc/bash_completion.d/ respectively. 

2. Our docker configurations are stored in the CARMAConfig repo. https://github.com/usdot-fhwa-stol/CARMAConfig. Please read the README contained there for an overview of that system. 

3. Once you have a config selected you can run the ./build-image.sh script located in the specific config folder you wish to use. This will create a usdotfhwastol/carma-config image. 

4. Once you have a config you can run
```
carma config install <config image name>
```
This will install the desired docker images needed for you carma configuration from our docker hub. If you wish yo build your own you may use the docker/build-image.sh script located in each repo to build the required images. These can then be pointed to by the docker-compose.yaml file in the carma-config image.

5. Make sure the following directories and files exist on your host machine
/opt/carma/logs
/opt/carma/.ros/
/opt/carma/maps/pcd_map.pcd
/opt/carma/maps/vector_map.osm
/opt/carma/vehicle (containing the calibration folder described in the readme of carma config). 

6. Run 
```
carma start all
```
7. To stop carma run 
```
carma stop all
```

TonysCousin, 2020-04-17:
Closing as apparent user error, since there has been no further contact in over 14 days.

closed 2020-04-17 by TonysCousin