usdot-fhwa-stol/carma-platform #632: Some nodes shut themselves down on SystemAlert of type FATAL instead of SHUTDOWN.

msmcconnell, 2020-04-06:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Some nodes are using an OR statement of FATAL and SHUTDOWN alert messages to determine when to shutdown. This is NOT desired behavior as it will result in undetermined nodes shutting down when a FATAL message is received even if the system could still function. 
### Expected behavior
Nodes only shut themselves down on receipt of a SHUTDOWN system alert message. 
### Software version this applies to

CARMASystem_3.3.0+
### Actual behavior
guidance_state_machine.cpp
status_camera.app (AVT Vimba Driver)
lidar_litev3hp_wrapper.cpp (Garmin Driver)
driver_application.cpp (CARMAUtils for most c++ drivers)
driver_wrapper.cpp (CARMAUtils for most c++ drivers)

All contain these OR statements as does
nm_application.cpp (XGV controller) (Low priority)

### Steps to reproduce the behavior

Run carma and publish system alert FATAL while using real software drivers. 

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-04-06:
I agree it is most correct to only shut down on a SHUTDOWN message.  However, for all practical purposes, a FATAL alert, by definition, indicates that we will be shutting down the system soon.  The term "fatal" is chosen because the system cannot continue to operate under such circumstances.  We need to be careful not to send FATAL alerts for any situation that is not truly fatal to the core mission of the system.  Having said that, it does make sense for nodes to shut down only in response to a SHUTDOWN message, and leave that decision to the health monitor.  Upon receiving a FATAL message, health monitor may choose to initiate some housekeeping before shutting everyone down.  Then, health monitor must be the only node that issues a SHUTDOWN message.

msmcconnell, 2020-04-06:
@TonysCousin I agree. However, I want to point out the use of FATAL has in the past been to indicate an error fatal to a signal node process and not the entire system (not clear if that was original design intent or just how it was implemented). If we wish to change that definition we should add another type to the system alert message to represent that distinction.