usdot-fhwa-stol/carma-platform #638: opendrive2lanelet2convertor missing units on speed limit

msmcconnell, 2020-04-07:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The open drive to lanelet2 converter script is populating the speed limit of the lanelets, but without units which is a requirement.
### Expected behavior
Tag
<key="limit" v="25 mph"/>
### Software version this applies to

CARMASystem_3.3.0
### Actual behavior
<key="limit" v="25"/>
### Steps to reproduce the behavior
Run opendrive2lanelet2converter.py on oxdr file and look at generated osm file. 
### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-04-20:
@msmcconnell what is the impact of not having this fix in place for in-lane cruising? I'm hoping the code will just parse out the numeric part of the field and if the units are not present, assume it is mph.

msmcconnell, 2020-05-05:
This was resolved by the linked PR and can be closed.

closed 2020-05-05 by msmcconnell