usdot-fhwa-stol/carma-platform #644: Shut down when runnig carma, on master branch

yomo11, 2020-04-14:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
This error occurs when start carma using

    $carma start all

of master branch.

    /usr/bin/carma: line 328: return: can only `return' from a function or sourced script
    Starting CARMA background processes...
    web-ui is up-to-date
    Starting CARMA Platform foreground processes...
    WARNING: The ORGANIZATION variable is not set. Defaulting to a blank string.
    WARNING: The PLATFORM_VERSION variable is not set. Defaulting to a blank string.
    WARNING: The SSC_VERSION variable is not set. Defaulting to a blank string.
    WARNING: The BASE_VERSION variable is not set. Defaulting to a blank string.
    WARNING: The LIGHTBAR_VERSION variable is not set. Defaulting to a blank string.
    WARNING: The COHDA_VERSION variable is not set. Defaulting to a blank string.
    WARNING: The VELODYNE_VERSION variable is not set. Defaulting to a blank string.
    WARNING: The VIMBA_VERSION variable is not set. Defaulting to a blank string.
    WARNING: The NOVATEL_VERSION variable is not set. Defaulting to a blank string.
    ERROR: no such image: carma-ssc-interface-wrapper:: invalid reference format
, and then carma shut down.

### Expected behavior
carma should running successfully.

### Software version this applies to

    Ubuntu 16.04
    Docker 18.09.7
    Docker-compose 1.10.0
### Actual behavior

### Steps to reproduce the behavior

1. Set carma configuration.

    $carma config set CARMASystem_urus-23-g92fe41b-lexus-rx-450h-2019

1. Start running carma

    $carma start all

### Related work

Link to related tickets or prior related work here.

Supraluminal, 2020-04-14:
Hi there! Thanks for the issue report, this is a good catch. We're in the process of changing up how some of our internal build systems work and this regression appears to be a result of that. We're presently investigating the issue, it looks like a file containing those environment variables currently isn't being baked into the docker image correctly and can hopefully push up a fix soon. In the mean time, I'd recommend backing out to an older commit on the CARMAConfig repo. Commit `f5cbb95f` should be prior to our new build-system being merged. 

We appreciate your patience as our new build systems are under construction! Thank you!

msmcconnell, 2020-04-15:
Issue was closed due to comments in referenced PR. Re-opening until verification of fix is complete.

msmcconnell, 2020-04-16:
@yomo11 Please let us know you still have this issue after pulling from develop on the CARMAConfig repo. If you experience another issue that is clearly unique from this one please open it in a separate issue thread.

TonysCousin, 2020-04-16:
Discussion w/ @Supraluminal reveals that this problem was on the develop branch, not on master (so not on an official release) of the CARMAConfig repo.  Further, this  can be worked around by building from source.  This is why I am reducing its severity from 1 to 2.

yomo11, 2020-04-17:
@msmcconnell This issue is already solved, thankyou very much. The `$carma config install` is working normally now, but something going wrong otherwise when `$carma start all`. Thence I opened a separate issue #649 .
Thank you!

closed 2020-04-17 by TonysCousin