usdot-fhwa-stol/carma-platform #645: Remove remaining Java code

TonysCousin, 2020-04-14:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
There is only one component remaining in the code base that is built from Java, the Message package.  Having Java code in the system provides several disadvantages that impede productive progress on further Carma developments.  These disadvantages include:
* Building the system requires a high-speed internet connection, as the Java compiler needs to access a gradle repository.  When at a test track with limited connectivity, this sometimes prevents making new builds in a timely way for integrating fixes.
* The Java code depends on use of rosjava, which is not actively maintained. This, in turn, locks us into a dependence on ROS Kinetic, which locks us into a dependence on Ubuntu 16.04.  Being locked into older versions prevents security updates and newer features from being adopted.
* Several rosbag operations are limited or prevented when using rosjava.
* There are some sophisticated Python tools available to analyze rosbags, but they are very difficult to get to work when rosjava is involved.  Having access to these tools would help streamline our integration testing and allow better analysis of performance.

### Expected behavior
All of Carma functionality is available without an Java code involved.

### Software version this applies to

3.3.0

### Actual behavior
Carma build includes Java code.

### Steps to reproduce the behavior
N/A

msmcconnell, 2020-05-04:
The Message package is not the only java node remaining. The mock drivers are as well, thought these would be much easier to transition to c++ given the limited functionality.

Fy2333, 2020-07-12:
@TonysCousin 
Hi Tony, can I ask you question. 

Have you successfully build Carma from source code in local? Actually, I am trying to build but it failed with ": Could not resolve all dependencies for configuration ':runtime'. Could not find org.ros.rosjava_messages: automotive_platform_msgs".  And also "Could not find metadata gov.dot.fhwa.saxton.carma:message/maven-metadata.xml in remote (file:/home/carma/carma_ws/carma/devel/share/maven/)" keep showing at the end of the building. 

I checked this error on "https://answers.ros.org/question/276858/rosjava-build-error-in-kinetic-could-not-find-orgrosrosjava_messagestf1119/"  it said that this error is relevant to the rosjave pkg, it may have some version conflict because of genjave. I tried to downgrade my genjave version with Synaptic Package Manager but failed with unknown reason. 

I have been stuck for few days. I will be grateful if you can offer some help.

msmcconnell, 2020-07-23:
@Fy2333 Please open a new issue with label (CARMA Support) that we can use to track your problem. Please avoid posting unrelated questions in existing issues. Build documentation can be found here https://usdot-carma.atlassian.net/wiki/spaces/CAR/pages/491651073/Using+a+CARMA+Platform+development+environment#Building-Code-and-Unit-Testing.

msmcconnell, 2021-03-01:
Resolved by above PRs

closed 2021-03-01 by msmcconnell