usdot-fhwa-stol/carma-platform #649: carma cannot run after built successfully

yomo11, 2020-04-17:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
some errors occured atfer

    $carma start all
and there are many lines of error message, so i tried

    $carma start all 2>&1 | tee carma_error.log
the log file as attachment.
[carma_error.log](https://github.com/usdot-fhwa-stol/carma-platform/files/4493683/carma_error.log)

But some of error message missing in the log file, including but not limited to

    roscore                    | IOError: [Errno 13] Permission denied: '/home/carma/.ros/roscore-11311.pid'
    ...
    lightbar-driver            | /usr/bin/du: cannot access '/home/carma/.ros/log': No such file or directory
    ...  
    avt-vimba-camera-driver    | /usr/bin/du: cannot access '/home/carma/.ros/log': No such file or directory
The above is part of the message.

[image]

### Expected behavior
CARMA should run successfully.
### Software version this applies to

    Ubuntu 16.04
    Docker 18.09.7
    Docker-compose 1.10.0

### Steps to reproduce the behavior

1. carma install and set
`$carma config install CARMASystem_urus-25-g34b9c53-lexus-rx-450h-2019`
`$carma config set CARMASystem_urus-25-g34b9c53-lexus-rx-450h-2019` 
1. Then try to run carma
`$carma start all`

Supraluminal, 2020-04-17:
Thanks again for reporting this, glad to see that your first issue seems resolved! Looks to me like it's not able to find the `/opt/carma/.ros` folder in the host environment (which gets mapped into Docker as the `/home/carma/.ros` you're seeing in your error messages). Can you confirm that folder exists on the host? If it does exist, can you check to make sure that user ID 1000 and/or group ID 1000 (you may need to look at your Linux system's user config to determine which user is 1000, normally it's the first non-root user created) can read that folder (as well as all of the `/opt/carma` directory structure)?

Thanks!

Supraluminal, 2020-05-01:
Closing this issue after no new discussion for 14 days. If the fix we suggested doesn't resolve the issue please feel free to let us know and we can re-open this and continue the investigation. For now not going to generate any further action items for our team as the current assessment is that the issue was a configuration problem rather than a discovered software defect.

closed 2020-05-01 by Supraluminal