usdot-fhwa-stol/carma-platform #662: Might be wrong datapath when carma config install

Helomair, 2020-04-21:
### Types of Issue

- [x ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
error occured after
```
$ carma config install
```

Here is error messages screenshot
[image]

The error messages
```
cat: can't open '/opt/carma/vehicle/config/docker-compose.yml': No such file or directory
cat: can't open '/opt/carma/vehicle/config/docker-compose-background.yml': No such file or directory
```
I've linked` /opt/carma/vehicle `to `carma-config/example_calibration_folder/vehicle`

and in `carma-config/example_calibration_folder/vehicle` only has `calibration` no `config`

### Expected behavior
config should install
### Software version this applies to

```
Ubuntu 16.04
Docker 18.09.7
Docker-compose 1.10.0
```
### Steps to reproduce the behavior
install configs
```
carma config install e871a3b-ford-escape-2010
```

Supraluminal, 2020-04-21:
Thanks for reporting this! I dug into it a little bit and have two things to note:

1) Looks like there is an error with our Ford Escape config, that didn't get update to our latest folder structure. It has the docker-compose.yml file in `/opt/carma/vehicle` instead of `/opt/carma/vehicle/config`. I'll plan to make a quick fix for that.  The Ford Escape platform isn't fully supported for us at this time, as it's our legacy platform but we can update this configuration error for sure. If you're just trying to use a config to test around with, I'd recommend either the Lexus, Pacifica, or Fusion configs as they are our current fleet. The development config will be best for actually running locally, as it uses mock-drivers instead of attempting to talk to hardware, but will require you to choose a CARMAPlatform docker image to tag as `latest` locally

2) Unsure if it's related to the issue you're experiencing or maybe I've misinterpreted but I'd recommend symlinking `carma-config/example_vehicle_calibration/vehicle/calibration` to `/opt/carma/vehicle/calibration`, rather than directly linking over `/opt/carma/vehicle`. It's unclear to me without some more testing whether or not that would pose problematic with the other items we map in via docker under `/opt/carma/vehicle`.

TonysCousin, 2020-04-23:
Marking this as sev4 since it applies to the notional Escape configs, a vehicle that we don't currently use.  Any vehicle included in our configs library should be represented correctly, but this particular one has minimal material impact.

Supraluminal, 2020-05-05:
Closing this issue after no new discussion for 14 days.

We've cleaned up the deprecated/unsupported configs in the carma-config repo, if you're looking to test in a development environment we recommend using the development config. If the fix we suggested doesn't resolve the issue (or if you encounter any issues with the supported configurations) please feel free to let us know and we can re-open this and continue the investigation. For now not going to generate any further action items for our team as the current assessment is that the issue was an unsupported configuration issue rather than a discovered software defect.

closed 2020-05-05 by Supraluminal