usdot-fhwa-stol/carma-platform #675: Fix Health Monitor for Trucks is not evaluating GPS failure only

maefromm, 2020-04-27:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

The Health monitor in develop is missing the logic or not evaluating the "s_1_l1_1_l2_1_g_0" which denotes GPS failure only. 

### Expected behavior
Should handle the GPS failure for trucks and unit testing should be added to check all of the combination of values. 

### Software version this applies to

### Actual behavior

### Steps to reproduce the behavior
1. Make sure to implement and test health monitor unit tests. 

### Related work
https://github.com/usdot-fhwa-stol/carma-platform/pull/639 
Link to related tickets or prior related work here.

TonysCousin, 2020-04-27:
@maefromm is it handling GPS failure in passenger cars?  What is the alert in that situation?  Should be a warning, I think.

maefromm, 2020-04-27:
@TonysCousin alert.type = cav_msgs::SystemAlert::CAUTION; to match the car logic that Ash had implemented.

TonysCousin, 2020-04-27:
Sounds good.  @maefromm are we ready to close this ticket?

maefromm, 2020-05-05:
Yes fix was done on the PR above.

closed 2020-05-05 by maefromm