usdot-fhwa-stol/carma-platform #678: UI rosbridge for ROS kinetic cannot launch properly

msmcconnell, 2020-04-27:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The Web UI cannot function because rosbridge is crashing on startup. This occurs on both the develop branch and a new local build of CARMASystem_3.3.0. The following error is thrown 
```
process[rosbridge_websocket-2]: started with pid [72]
2020-04-27 19:08:22+0000 [-] Log opened.
2020-04-27 19:08:22+0000 [-] Traceback (most recent call last):
2020-04-27 19:08:22+0000 [-]   File "/opt/ros/kinetic/lib/rosbridge_server/rosbridge_websocket", line 48, in <module>
2020-04-27 19:08:22+0000 [-]     from rosbridge_server import ClientManager
2020-04-27 19:08:22+0000 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/__init__.py", line 2, in <module>
2020-04-27 19:08:22+0000 [-]     from .websocket_handler import RosbridgeWebSocket
2020-04-27 19:08:22+0000 [-]   File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_server/websocket_handler.py", line 43, in <module>
2020-04-27 19:08:22+0000 [-]     from tornado import version_info as tornado_version_info
2020-04-27 19:08:22+0000 [-] ImportError: No module named tornado
```

It appears that the python module tornado cannot be found.
### Expected behavior
rosbridge can be launched normally. 
### Software version this applies to

CARMASystem_3.3.0
### Actual behavior
The above crash appears and the UI cannot connect to ROS. 
### Steps to reproduce the behavior

On a build of develop or clean build of any previous version run carma start all. Or connect to a container and run ```roslaunch rosbridge_server rosbridge_websocket.launch```

### Related work

It appears this issue might be related to a faulty binary from 
```sudo apt-get install ros-kinetic-rosbridge-server```
This would make sense given the fact that it breaks on older versions of carma. 
Therefore, the following issue has been opened on the rosbridge repo https://github.com/RobotWebTools/rosbridge_suite/issues/490

msmcconnell, 2020-04-27:
Looks like there was another issue on the same repo posted. They have reverted the root cause changes, but we need to wait for the next kinetic sync before they are applied. 
https://github.com/RobotWebTools/rosbridge_suite/issues/484

msmcconnell, 2020-04-27:
Kinetic sync's occur every 2 weeks. The next one would be around the 31st. 
https://discourse.ros.org/c/release/kinetic/17

TonysCousin, 2020-04-27:
I would like to wait and see if their next release fixes it, unless this is preventing our current work from moving forward

msmcconnell, 2020-05-01:
Based on https://github.com/RobotWebTools/rosbridge_suite/issues/484#issuecomment-621366237
the bug fix has been released. And based on https://discourse.ros.org/t/preparing-for-kinetic-sync-2020-04-29/13864 and http://repositories.ros.org/status_page/ros_kinetic_default.html?q=SYNC

The kinetic sync will occur on May 2nd and it looks like the new version 0.11.6 is included. So this should be fixed by Monday.

msmcconnell, 2020-05-04:
Fix should be available with apt-get now. https://discourse.ros.org/t/preparing-for-kinetic-sync-2020-04-29/13864

We need to rebuild to confirm.

dan-du-car, 2020-05-04:
Verified with carma-messenger images. It works properly. 
1) clone carma-base, carma-messenger repo
2) run ./build_image.sh -d located in carma-base/ docker folder. Then build images for carma-messenger. Result below: 
[image]

3) run "carma config set ... "  and "carma start all". Rosbridge is running without issue. See below:
[image]

maefromm, 2020-05-21:
@dan-du-car @msmcconnell I have validated that the carma-messenger still having this issue even with the new carma-base build for carma-messenger ui. Reopening this ticket. 

[image]

maefromm, 2020-05-21:
1) Ensured https://hub.docker.com/repository/docker/usdotfhwastoldev/carma-base was rebuilt by JB. 

2) Remove all your docker images, containers and volumes to ensure everythings removed. 

carma stop all
docker system prune -a 
docker system prune -a --volumes
docker images ls
docker container ls
docker volumes ls

3) Pull and rebuild all images. 

docker pull usdotfhwastoldev/carma-base:develop
docker images
cd carma-messenger-core/docker/
./build-image.sh -d
docker images 
cd carma-messenger-ui/docker/
./build-image.sh -d
docker images
cd ../carma-messenger-config/development/development/
./build-image.sh -d
docker images

sudo service apache2 stop
carma config set usdotfhwastol/carma-messenger-config:e49bf2c-development 
carma start all

Then error showed up.

maefromm, 2020-05-26:
JB the carma-base image seems to have been fixed. Re-did the steps above and was able to run. 
[image]

However, the carma-messenger-config is bad. Somehow pointing to the usdotfhwastol instead of dev. Rebuilding it locally using the steps above worked. Will probably need to rerun all 3 carma-messenger images without the cache option. 
[image]

msmcconnell, 2020-05-27:
@maefromm The carma-config issue seems separate from this. Might be a good idea to close this and track it in a new ticket.

maefromm, 2020-05-27:
I agree I'll close it since it's not related.

closed 2020-05-27 by maefromm