usdot-fhwa-stol/carma-platform #679: CARMA Messenger Docker Image needs to include the Truck Inspection Plugin

maefromm, 2020-04-27:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Found that the Truck Inspection Plugin was not being included by the carma-messenger docker build-image.sh and had to be launched manually during the local integration testing. Currently anything under the carma-messenger-core and the ui are being built into images. 

Need to move the truck_inspection_plugin under the carma-messenger-core folder and docker scripts updated accordingly. 

### Expected behavior
The truck inspection plugin should be included in the carma-messenger image. 

### Software version this applies to

develop

### Actual behavior
Had to be manually launched. 

### Steps to reproduce the behavior
1. Build the image to ensure the truck inspection plugin can be started using docker and without having to execute the launch file manually. 

2. Quick local integration tested for truck inspection plugin and ui. 

### Related work
Link to related tickets or prior related work here.

maefromm, 2020-04-27:
@TonysCousin this is higher than severity 3 in terms of carma-messenger to work properly.

dan-du-car, 2020-05-04:
The changes are merged in carma messenger repo referencing Pull Request #11

closed 2020-05-04 by dan-du-car