usdot-fhwa-stol/carma-platform #690: Investigation of unknown opendrive2lanelt2 converter bugs

itabrl, 2020-05-05:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
opendrive2lanelet2converter has some unknown bugs that results in marking the lanelets left and right edges incorrectly it is unknown that if this error is part of errors in script or in the third party or in the original opendrive map. this needs to be investigated more deeply. 

### Expected behavior
correct lanelets with correct left and right edges. 

### Software version this applies to

### Actual behavior

### Steps to reproduce the behavior

Convert ATEF.xodr map using the converter

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-05-05:
Assigned sev2 based on discussion w/Samir.  This could happen when converting any given map, and the results could prevent Carma Platform from using that map, so it's an essential function.  But there is a workaround in that the output map (Lanelet2 format) can be manually edited to be usable.

msmcconnell, 2020-05-08:
Per discussion with @itabrl. This issue does not relate to a single bug but rather the collection of bugs we have found in the past, which are relevant usually to specific maps. Since the source of the bug is not known to be the OpenDRIVE map or the converter further investigation is needed. Some concrete issues we discussed were
* Some lanelets not being matched to the correct lane edge
* Turn direction assignment not being correct
* FLETC geometry issues

Since these have been resolved manually in the past without separate github issues their isn't a perfect record of which bugs have been encountered before so the above list is not necessarily comprehensive. Further discussion is needed on a general approach to map validation CAR-2468 is a good starting point.

TonysCousin, 2020-05-08:
Closing as invalid, since it has been replaced by a more specific description in #696 .

closed 2020-05-08 by TonysCousin