usdot-fhwa-stol/carma-platform #696: ATEF map cannot be routed over without manual edits.

msmcconnell, 2020-05-08:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
When the [opendrive2lanelet converter](https://github.com/usdot-fhwa-stol/opendrive2lanelet) is used on the OpenDRIVE ATEF map the resulting lanelet OSM map is not rotatable and has an incomplete routing graph. It is unclear if the problem arises from the base OpenDRIVE map or the converter. 
@itabrl was able to make manual edits to correct the lanelet edge id and lanelet id associations to resolve this but an automated fix or base map fix has not been developed. 
### Expected behavior
ATEF map is free from issue preventing routing and usage in carma after passing through converter without additional manual edits. 
### Software version this applies to

### Actual behavior
ATEF map cannot be used in carma without manual edits. 
### Steps to reproduce the behavior
Run the original ATEF XODR file through the converter and attempt to use the osm file or look at the routing graph generated by lanelet. 

### Related work
#690 
Link to related tickets or prior related work here.

TonysCousin, 2020-05-08:
Severity 2 for reasons documented in #690 .

msmcconnell, 2020-05-27:
This was resolved by the above PR

closed 2020-05-27 by msmcconnell