usdot-fhwa-stol/carma-platform #700: findReference is not returning certain valid references

MishkaMN, 2020-05-14:
### Types of Issue

- [X] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Recently added findReference function is not accounting for the fact that linestrings can be owned by regulatory elements (regem) aside from lanelets and areas. There is an edge case where unregistered primitives, such as a linestring that is owned by a certain regem, would not return the parent regem which is registered in the map. The function would fail to find that relation.

### Expected behavior
The function should return items that are referenced by items that are not directly registered in the map.  

### Software version this applies to

On carma-develop branch of autoware.ai soon after 3.3.0

### Actual behavior
See above.

### Steps to reproduce the behavior
Pass a linestring to the function that is owned by a regem, which is registered in the map. The function does not return that regem. 
// Pseudocode
Linestring ls
RegulatoryElemetPtr regem_ptr(ls)
map.add(regem_ptr)
ref = findReference(ls, map)
// ref obj does not have regem_ptr 

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-05-14:
@MishkaMN it's not clear what the impact of this problem is, i.e. how severely does it affect Carma operation.  Can you please add that to the description.

MishkaMN, 2020-05-14:
The fix is merged in with the linked PR.

msmcconnell, 2020-05-15:
Closing as resolved by linked PR 103

closed 2020-05-15 by msmcconnell