usdot-fhwa-stol/carma-platform #733: RVIZ Segmentation Fault via carma exec

kjrush, 2020-06-23:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
When attempting to run rviz via CARMA exec script the following segfault is encountered:

```$ carma exec rviz -d /opt/carma/install/share/carma/rviz/carma_default.rviz
Trying to open shell inside image: usdotfhwastoldev/carma-platform:develop
[ INFO] [1592785861.151249069]: rviz version 1.12.17
[ INFO] [1592785861.151503265]: compiled against Qt version 5.5.1
[ INFO] [1592785861.151532438]: compiled against OGRE version 1.9.0 (Ghadamon)
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/bin/bash: line 1:   116 Segmentation fault      (core dumped) rviz -d
/opt/carma/install/share/carma/rviz/carma_default.rviz
```
### Expected behavior
RVIZ loads normally and displays the requested RVIZ file.
### Software version this applies to
3.3.0 

### Actual behavior
RVIZ segementation faults
### Steps to reproduce the behavior

1. Run `carma exec rviz -d /opt/carma/install/share/carma/rviz/carma_default.rviz`

### Related work

Link to related tickets or prior related work here.

msmcconnell, 2020-06-29:
Confirmed that this is repeatable on Blue Lexus PC. Can't reproduce in virtual machine so it is likely related to how displays are managed differently between both configurations
```
dev@dev-Nuvo-6108GC:~$ carma exec rviz
Trying to open shell inside image: usdotfhwastoldev/carma-platform:develop
[ INFO] [1593454534.267866317]: rviz version 1.12.17
[ INFO] [1593454534.267911934]: compiled against Qt version 5.5.1
[ INFO] [1593454534.267922703]: compiled against OGRE version 1.9.0 (Ghadamon)
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Could not initialize OpenGL for RasterGLSurface, reverting to RasterSurface.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
/bin/bash: line 1:   116 Segmentation fault      (core dumped) rviz
```

msmcconnell, 2020-08-07:
@kjrush A fix was merged for this, I think we need to follow up with the original reporter to see if their issue is fixed.

msmcconnell, 2020-09-01:
Following up on what we believe the solution to be. Note this only works with Nvidia GPUs
Step 1. Install Nvidia Docker Runtime[1]
```
distribution=$(. /etc/os-release;echo $ID$VERSION_ID)
curl -s -L https://nvidia.github.io/nvidia-docker/gpgkey | sudo apt-key add -
curl -s -L https://nvidia.github.io/nvidia-docker/$distribution/nvidia-docker.list | sudo tee /etc/apt/sources.list.d/nvidia-docker.list

sudo apt-get update && sudo apt-get install -y nvidia-container-toolkit
sudo systemctl restart docker
sudo apt-get install nvidia-container-runtime

sudo tee /etc/docker/daemon.json <<EOF
{
    "runtimes": {
        "nvidia": {
            "path": "/usr/bin/nvidia-container-runtime",
            "runtimeArgs": []
        }
    }
}
EOF
sudo pkill -SIGHUP dockerd
sudo systemctl daemon-reload
sudo systemctl restart docker
```
Then try the rviz command again
```
carma exec rviz
```