usdot-fhwa-stol/carma-platform #743: AVT Vimba Camera Driver cannot find shutdown_node

msmcconnell, 2020-07-08:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
When launching the carma develop config for the lexus. The AVT driver container throws the following error
```
avt-vimba-camera-driver    | ERROR: cannot launch node of type [driver_shutdown/driver_shutdown]: driver_shutdown
avt-vimba-camera-driver    | ROS path [0]=/opt/ros/kinetic/share/ros
avt-vimba-camera-driver    | ROS path [1]=/opt/carma/install/share
avt-vimba-camera-driver    | ROS path [2]=/opt/ros/kinetic/share
```
It appears that the driver_shutdown node used for clean system shutdown cannot be found. Most likely this is because the checkout.sh script is not pulling in the most up to date dependency when the image is built.
### Expected behavior
No startup errors for any image.
### Software version this applies to

develop post 3.3.0 release on commit https://github.com/usdot-fhwa-stol/carma-platform/commit/b39d3956c533cb99e196e8b5f8af320d10cb6090

AVT Vimba commit https://github.com/usdot-fhwa-stol/avt_vimba_camera/commit/0381f36ae11e905318b7729b59b895668a692810
### Actual behavior
AVT Vimba Camera Docker image throws the above error on start up
### Steps to reproduce the behavior

1. Pull dev images for lexus config and set the config
2. run carma start all

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-07-08:
@msmcconnell please confirm impact of this problem.  It sounds like the camera driver will not start, but it is the only image impacted, so the remainder of Carma is starting okay.  Is that a correct assessment?

msmcconnell, 2020-07-08:
@TonysCousin Yes that appears to be correct.

msmcconnell, 2020-07-30:
Resolution of this is functionally blocked by #787 as new camera docker images cannot be deployed to dockerhub

msmcconnell, 2020-07-31:
Resolved by above PR

closed 2020-07-31 by msmcconnell