usdot-fhwa-stol/carma-platform #744: Ford Fusion Drive By Wire message type mismatch

msmcconnell, 2020-07-08:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The ford fusion SSC on develop cannot run because the drive by wire system messages have been updated to a newer version
```
ssc_controller_driver      | [ERROR] [1594221723.589642311]: Client [/hardware_interface/veh_interface] wants topic /hardware_interface/ds_fusion/misc_1_report to have datatype/md5sum [dbw_mkz_msgs/Misc1Report/036c26779fee8ddc637f32d336e03b66], but our version has [dbw_mkz_msgs/Misc1Report/551ea602464aaab22bc1752e844dde30]. Dropping connection.
```
### Expected behavior
SSC can run on develop build
### Software version this applies to

develop post 3.3.0
### Actual behavior
The above error appears after running carma start all
### Steps to reproduce the behavior
1. Pull and install the develop config for the ford fusion
2. Run carma start all
### Related work

The cause of this issue is likely similar to this previously resolved issue https://github.com/usdot-fhwa-stol/carma-platform/issues/370

That issue was fixed by getting new SSC binaries. I think another solution might be to lock the dependency to a specific version in the checkout.sh script.

TonysCousin, 2020-07-08:
Since this anomaly takes the Fusion out of service completely, it is a severity 1 issue.  There is no workaround (that can be employed by an end-user).  It appears the only way around this is to either roll back the controller driver code, or update Carma code to use the new message format, which would result in a new Carma software version.  Either way, I would like to lock our checkout.sh to be pulling from a specific commit (preferably from a tag) in the New Eagle repo, rather than getting the head of any branch there.  This should be a common practice on anything we pull in from a third party.

TonysCousin, 2020-07-30:
We are waiting on AStuff for a new build of SSC in order to correct this.

msmcconnell, 2020-08-26:
This is currently being worked by AutonomouStuff as part of ticket 3204. It was previously ticket 3203 but merged in with the another SSC ticket.

jtbaird, 2020-09-23:
fixed by above PRs

closed 2020-09-23 by jtbaird

maefromm, 2020-09-28:
@TonysCousin I believe we discussed with @msmcconnell  and Sudhakar that we agree that AStuff should be packaging the fixes for us instead of us doing the compilation into our own image ourselves. Correct?