usdot-fhwa-stol/carma-platform #745: RSU firmware not accepting message forwarding configurations

jtbaird, 2020-07-08:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
When trying to set up OBUs for message forwarding, snmpset commands are not working as expected and so we are not able to set up message forwarding.

### Expected behavior
snmpset commands in [OBU_Upgrade.txt](https://vector.leidos.com/sites/STOL/STOL2/Shared%20Documents/Current%20Projects/TO%2018-225%20Carma3/Resources/Installation%20and%20Manuals%20and%20Training/OBU/OBU_Upgrade.txt) should successfully set up message forwarding
### Software version this applies to

RSU firmware 68899
### Actual behavior
snmpset commands do not modify MIB but instead return error "command not found in this MIB"
### Steps to reproduce the behavior
Run commands in above OBU_Upgrade.txt file

### Related work

Link to related tickets or prior related work here.

abey-yoseph, 2020-07-13:
Hi, I heard there was a bit of confusion about this. To enable message forwarding on the Cohda OBUs, several SNMP OIDs must be updated (Steps 6-11 in Part 2 of the OBU_Upgrade.txt). When we try to do this, the OBUs throw an error saying that the OIDs are not contained in the MIB. We have tried switching between several different versions of the firmware, old and new, as the MIB could have been altered in firmware upgrades. None of the firmware versions we tried proved successful. I've tried reaching out to Cohda with this issue, but have not heard back unfortunately.

jtbaird, 2020-07-14:
Issue has been resolved by using new steps from Cohda which can be found [here.](https://vector.leidos.com/sites/STOL/STOL2/Shared%20Documents/Current%20Projects/TO%2018-225%20Carma3/Resources/Installation%20and%20Manuals%20and%20Training/OBU/Setup%20Message%20Forwarding.txt)

closed 2020-07-14 by jtbaird