usdot-fhwa-stol/carma-platform #747: SSC Version out of date with pacmod_msgs

msmcconnell, 2020-07-09:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The SSC version on the blue lexus is 1.0.1 which is incompatible with the newest pacmod_msgs version. 
The newest version of the lexus SSC is 1.1.1.

The suggested fix is to upgrade the SSC binaries. This can be done with 
sudo apt-get install ros-kinetic-ssc-pm-lexus
However, this requires that the PC in use be one with AStuff installed access keys which is documented in the Leidos vector under the CARMA 3 Resources / Installation Manuals and Training / SSC / SSC License and Management.docx

### Expected behavior
SSC is compatible with pacmod version installed in carma-base. 

### Software version this applies to

develop after 3.3.0

### Actual behavior
Checksum error occurs between differing message versions of SSC and pacmod. 

### Steps to reproduce the behavior

1. Run develop build of carma
2. Look at error logs from ssc. 

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-07-09:
Marking this as sev 3 since we can use an older version of carma-base image to get around this for the time being.

closed 2020-07-20 by kjrush

msmcconnell, 2020-07-20:
Closed by above PR