usdot-fhwa-stol/carma-platform #749: Update j2735_convertor for new geofence message spec - Control Message

afrye-leidos, 2020-07-10:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary

Carma cloud team recently changed their definition of geofence. The new spec is under
https://github.com/usdot-fhwa-stol/carma-cloud/blob/master/cc_traffic_control_messages_asn1.txt
A subset of the new messages were created and merged in https://github.com/usdot-fhwa-stol/carma-msgs/pull/58

Now, these new messages need to be added to the j2735_convertor in carma-messenger. 

### Expected behavior

All new messages under and including the TrafficControlMessage have a convertor from cav_msgs to j2735_msgs and back, and each of these is unit tested. 

### Software version this applies to

carma-msgs integration/geofence_msg

### Actual behavior

There are no convertors or unit tests for these new messages

### Steps to reproduce the behavior

N/A

### Related work

Follow-on for https://github.com/usdot-fhwa-stol/carma-platform/issues/748

MishkaMN, 2020-07-13:
Resolved by above merge

closed 2020-07-13 by MishkaMN

MishkaMN, 2020-07-13:
Made a small fix to the converter by above PR