usdot-fhwa-stol/carma-platform #756: Blue Lexus cannot use NAS logging

msmcconnell, 2020-07-16:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Blue lexus cannot log to the NAS data storage unit. Seems to be a permissions issue. 

### Expected behavior
Blue lexus will automatically be able to log to the NAS at startup and carma will log there by default
### Software version this applies to

develop after 3.3.0
### Actual behavior
Blue lexus cannot log to the NAS data storage unit. Seems to be a permissions issue. 
### Steps to reproduce the behavior

Run carma with carma start all

### Related work

May or may not be related to https://github.com/usdot-fhwa-stol/carma-platform/issues/609

msmcconnell, 2020-07-16:
@snallamothu Please add any clarifications you can provide

jtbaird, 2020-09-02:
Resolved, mount command in /etc/rc.local was pointing to the wrong IP.

closed 2020-09-02 by jtbaird