usdot-fhwa-stol/carma-platform #758: Platooning strategic plugin cannot be built with docker build

msmcconnell, 2020-07-20:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
When trying to build develop docker images the following error occurs
```
[ 85%] Generating C++ code from cav_srvs/SetLightBarIndicator.srv
[ 85%] Generating C++ code from cav_srvs/GetAvailableRoutes.srv
In file included from /home/carma/carma_ws/src/carma-platform/platooning_strategic/include/platoon_strategic.hpp:36:0,
                 from /home/carma/carma_ws/src/carma-platform/platooning_strategic/src/platoon_strategic.cpp:19:
/home/carma/carma_ws/src/carma-platform/platooning_strategic/include/state_machine.hpp:8:36: fatal error: cav_srvs/PlanManeuvers.h: No such file or directory
compilation terminated.
carma-platform/platooning_strategic/CMakeFiles/platoon_strategic_plugin_lib.dir/build.make:62: recipe for target 'carma-platform/platooning_strategic/CMakeFiles/platoon_strategic_plugin_lib.dir/src/platoon_strategic.cpp.o' failed
make[2]: *** [carma-platform/platooning_strategic/CMakeFiles/platoon_strategic_plugin_lib.dir/src/platoon_strategic.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```
### Expected behavior
No build errors on develop branch
### Software version this applies to

develop after version 3.3.0
Commit: https://github.com/usdot-fhwa-stol/carma-platform/commit/9a6b25b997463d27a0929e6414a684089a6ff9f7
### Actual behavior
Above error occurs
### Steps to reproduce the behavior
checkout develop
./build-image.sh -d

### Related work

Link to related tickets or prior related work here.

TonysCousin, 2020-07-20:
Since this prevents our key Docker image from building it is sev 1.

msmcconnell, 2020-07-20:
Resolved by above PR

PR #759 (2020-07-20): Add missing dependancies for cav_srvs

# PR Details
## Description

Add the missing cav_srvs dependency to the platooning_strategic plugin package. 
## Related Issue
Resolves #758 

## Motivation and Context
Keep develop buildable

## How Has This Been Tested?
Local docker build using build-image script and CI

## Types of changes

- [x] Defect fix (non-breaking change that fixes an issue)
- [ ] New feature (non-breaking change that adds functionality)
- [ ] Breaking change (fix or feature that cause existing functionality to change)

## Checklist:
closed 2020-07-20 by msmcconnell