usdot-fhwa-stol/carma-platform #761: Guidance "get_registered_plugins" and "get_active_plugins" does not show the required plugins

maefromm, 2020-07-22:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During the UI implementation for Basic Travel using the integration/routing branch of carma-platform, we found that the following topics displayed the wrong list of plugins: 

/guidance/plugins/get_registered_plugins 
/guidance/plugins/get_active_plugins 

### Expected behavior
Should show at least 1 plugin for Strategic, Tactical and Control plugin types. Except for Platooning was expecting as shown on the attachment below. 
[image]

### Software version this applies to

INTEGRATION/ROUTING branch

### Actual behavior
See attached. 
[image]

### Steps to reproduce the behavior
1. Build integration/routing branch of carma-msgs and carma-platform. 
2. run carma start all. 
3. echo the topics above. 

### Related work
Link to related tickets or prior related work here.
https://usdot-carma.atlassian.net/browse/CAR-2512

maefromm, 2020-07-22:
Per @kjrush the Autoware Plugin should be updated to type = Tactical and the name = In-Lane Cruising
[image]

As for Route Following and Pure Pursuit plugins, need further research. 

@dan-du-car needs these topics to display and not able to fake the data since it is a service call.

TonysCousin, 2020-07-22:
Marking sev 1 since it appears that plugins cannot be reliably selected.

TonysCousin, 2020-07-30:
Discussion w/Mae, Kyle, Mike resolve that this is sev 3 because the effect is that the plugin list won't display, and individual plugins won't be able to be deactivated (via UI), but we expect that the UI will otherwise work and allow the vehicle to drive.

closed 2020-08-13 by maefromm